Chris-Annin/AR2 #1: 3D cad file

quangphuc91, 2017-07-23:
Very nice project

Can you send me step file , because my 3d printer can't printing more than 200mm, i need step file to split more part have dimension less than 200 mm

Thanks you

oconnorhorrill, 2017-07-23:
Hi Chris,

think you may have send this email to the wrong address. The AR2 is a great
project.

Kind regards,

Jim O’Connorhorrill.

Chris-Annin, 2017-07-24:
im sorry I ddint get a chance to reply - busy weekend.  I have had many
people ask for the step files - I havnt posted them yet as when we started
bulding the aluminum version of this robot we found some unexpected
differences in the bearing models.  I found the same issues in my plastic
prints and I made adjustements to the models I was using for printing but I
have not yet made any adjustments to the models I sent my machinist as he
is confirming exact changes that need to be made - so basically what im
saying is I have 2 different sets of models, a set for the aluminum version
and a set for the printed version.  Ive been telling everyone they need to
wait for me to finish working the bugs out on the aluminum models before I
can share them.  If you only need models for 3D printing I can sent you
what I have - just understand they wont be perfect if you try to use them
for machining as I have made many little adjustments to get them to print
out perfectly.   I will email you the models when I get home from work
tonight.  Its been a very hectic day so feel free to send me a reminder
email tonight if I dont get to it right away.

Chris

Chris-Annin, 2017-07-25:
will .sldprt files work?  I haven't converted them to step files yet.  message me at chris.annin@gmail.com

quangphuc91, 2017-07-25:
yes , can you send me step file , b/c im using solidwork 2014 sp5

alfredanil, 2017-07-25:
Would be helpfull if the step files or solidworks files are uploaded/added to the main project.

quangphuc91, 2017-07-25:
thank you very much

Chris-Annin, 2017-08-03:
> Mobile: (+84)935294991 <+84%2093%20529%2049%2091>

step files are now posted on the project page.

Chris-Annin, 2017-08-03:
step files are now posted on the project page.

aaron02, 2017-10-12:
can i use the current files to made on of these out of aluminium ? or is this still in progress?

great roject :)

Chris-Annin, 2017-10-13:
yes the step files are good to go.  also read the notes on page 2 of the assembly manual.

conor87, 2017-12-11:
Can you publish Assembly file in step?
Project is brilliant and description is very well but with assy will be easier to build it.

Chris-Annin, 2017-12-11:
hellow,  just got the solidworks files posted a few days ago.  I can work
on rebuilding the assembly.  it got a little messy and I wanted to clean it
up.  I'll see what I can do to get the solidworks addembly together.

Chris

conor87, 2017-12-11:
This files are not compatible with solidworks 2015 (propably newer version), it's why I ask for step. If you are in assy enviroment, just save it as step and job is done :-)

conor87, 2017-12-11:
Question:
how it works with Python?
I love this language but since few weeks I try to work in Java. Main reason -> javafx. In Python GUI is still in the background.
Did you have any problems?

photoJames, 2018-03-19:
Couldn't find any files in the projects table. Would love to 3d print for the STEM lab I teach.

Chris-Annin, 2018-03-20:
3D print files are posted "3D print files.zip"   please let me know if any issues.

closed 2018-03-20 by Chris-Annin

soswow, 2018-04-07:
> step files are now posted on the project page.
Where are those?