Chris-Annin/AR2 #3: How are you reducing play in the planetary gearboxes?

i-make-robots, 2017-11-19:
I received gearboxes from omc-stepperonline.com that claim 1 degree of play.  in practice it feels like more, but that might be my printer tolerances on this similar robot:

Chris-Annin, 2017-11-20:
I agree,  Ive also found is slightly beyond 1°.  if opposing force is applied to the arm there is some backlash but with the robot under load or compression I have not had any issues and the robot repeats to position very well.   At some point in the future when I have more time I may look at designing the next generation/next size up robot with more expensive gearboxes (it will be substatially more - likely 5x to 7x price) but for now given size and budget I felt these motor/gearboxes were the most economical option I could find.

mdjurfeldt, 2017-11-23:
Could you estimate the position repeatability (in mm) for the plastic version and the aluminum version?

Chris-Annin, 2017-11-23:
Approx .075 to 1.5mm is what I have experienced in testing where I ran an exercise program several times and brought the robot back to a dial indicator to check repeatability.

Chris-Annin, 2017-11-24:
> Thanks. And you dont't mean 0.075 to 0.15mm?

sorry for the delay.  I get alot of emails on the robot, difficult to keep
up.  .75mm is a good expectation - from what testing ive done.  keep in
mind these are very low cost gear boxes.  I think if I build next version
with high end gear boxes we could get alot better repeatablity.  but .75
isnt too bad for an inexpensive material handling unit.

i-make-robots, 2017-11-24:
I found that the plastic parts I designed get chewed up at the motor shaft because they can't take the torque.  I'm currently using FEA to design some steel mounting adapters to relieve the strain.  How did you solve/avoid this issue?

Chris-Annin, 2017-11-25:
are you having trouble with parts for the AR2 robot? or are you saying you have other parts that are having issues?

on my robot I print any spindles that have key shafts requiring torque out of ABS at 100% solid, all spindles are pressed into or captured around the OD by a taper bearing - the steel bearing keeps the ABS compressed and doesn't allow it to displace or break apart and then lastly I make sure to buy key-stock the runs the entire length of the spindle to distribute load as far across the ABS as possible rather that a short piece of key. 

thank you for the complement on the J5 drive - difficult to come up with low cost solutions to transfer motion 90° with the motor mounted at a distance without an expensive gear box,  It took me a couple days of contemplating before that one came to me.  Im thinking of using similar process for more joints on a larger scale for my next robot.

i-make-robots, 2017-11-25:
I would love to collaborate on a design.  My plastic problems are in this machine: https://www.instagram.com/p/Ba68yt6AKv8
which I have since updated to have no belts, no glue, and (at present) no metal parts that must be custom fabricated.  https://www.instagram.com/p/Bbvp4Ijgzs8/
The keys at the elbow and shoulder bite through the ABS, but I only printed it at 20% infill.  I will try it again at 100% and see if that works better.  My next option is a steel disc that couples the plastic to the shaft with bolts via six tapped holes on the disc.

i-make-robots, 2017-11-25:
It occurs to me that your J5 doesn't even need a closed loop belt, which will make sourcing and tensioning easier (maybe?)

Kristong, 2017-12-07:
I found that "J5 main bearing housing" in SolidWork files, the dimension is not correct !

Chris-Annin, 2017-12-09:
which dimension do you think is incorrect?

Kristong, 2017-12-09:
I'm do assembly your AR2 by your Solidworks files.
I found 2 parts ..
J5 main bearing housing.
J6 main bearing arm.

After I use J6 main bearing it correct., OK.
I don't know where J5 main bearing housing
, where to assembly. .?

Best regards,
Kristong S.

closed 2018-03-20 by Chris-Annin