Chris-Annin/AR2 #5: Missing screw holes in Base support part 1 and 2

Johansson-D, 2017-12-09:
Hi Chris_Annin.
I just joined this "little" project. :)
The M4 holes in the Base support part 1 and 2  are missing, or should one drill them?
The M4 holes are shown in the BOM pdf on page 3.

File names:
J1 base support spacer part 1 2017.09.18
J1 base support spacer part 2 2017.09.18

Best regards and THAN YOU for sharing this/
Dahniel

Chris-Annin, 2017-12-09:
hello, I was thinking perhaps they were not needed and just added extra
work and cost so I removed them from the models.  if you feel they are
needed I could put them back but I thought the 4 main mounting holes would
be sufficient but let me know what you think.

Chris

Johansson-D, 2017-12-10:
Hi Chris,
thanks for a quick reply.
Are you saying that the M4 screw are not needed?
I would be glad if you could mail the files or put them up at the website
again. Because then one has the option of chosing.

Is this a hobby for you Chris, or do you work with robots?
I just started a two year master in robotics and automation and I love it.
Example of courses we have had so far are:
- Robot modeling using Matlab
- Robot simulation using ABB:s Robotstudio
- Robotsystem where we play with the real deal, seven ABB robots. 😀

Best regards,
Dahniel Johansson

Johansson-D, 2017-12-16:
Hi Chris.
Did you put up the files somewhere?
I feel that they are needed.
Best regards,
Dahniel

Chris-Annin, 2017-12-16:
I believe if you go to Code, then click on commits and go to may 29th 2017 and click view repository at this point in time you should find the earlier print files.

https://github.com/Chris-Annin/AR2/tree/20a7b994f84574137bca8c4c3074bd3974aa6010

Johansson-D, 2017-12-22:
I found them here:
https://github.com/Chris-Annin/AR2/blob/fe274c6ada36cc75682504b6163ea55a67c506cc/AR2%20print%20files.zip

Thx Chris

Chris-Annin, 2018-03-20:
updated design calls for (x2) 4mm dowels rather than wasted expense on many screws.

closed 2018-03-20 by Chris-Annin