Chris-Annin/AR2 #7: Could you add more comments in sketch_Annin_Robot.ino ?

cqliliping, 2018-01-09:
Hi.Chris_Annin.
I'm reading  arduino code,but it is too difficult to me to comprehend somewhere in short time.especially in code line 831 ////CALC SPEEDS///// ,/////DRIVE MOTORS/////. I want to know how it works,then I can make every stepper works on the same speed with different pulse.
I'm from China and interesting in this project,and learning it.I just hope everyone can understand it rapidly and practice.
Best regards
liliping

Chris-Annin, 2018-01-10:
hi.  below is my algorithm.  first i find which motor has the most steps
and that is called high step, then I divide each of the other motor steps
into high step to figure out which loops of the while statement to pulse
the motor on.  the motor with highest step pulses the motor every loop -
the other motors will only pulse on certain loops based on the divide.  it
will not divide evenly there is always left over so I take the remainder
and then run a calculation to spread these leftovers evenly across the span
of high step and then I skip these loops, and then I do it again to catch
any more leftovers and create a 3rd level of skips.  its confusing but you
will have to study the code below.  PE stands for Pulse Every (loop) and SE
stands for Skip Every (loop).  I will try to add more comments next time i
make a rev change.  thank you.

        //FIND HIGHEST STEP
        int HighStep = J1step;
        if (J2step > HighStep)
        {
          HighStep = J2step;
        }
        if (J3step > HighStep)
        {
          HighStep = J3step;
        }
        if (J4step > HighStep)
        {
          HighStep = J4step;
        }
        if (J5step > HighStep)
        {
          HighStep = J5step;
        }
        if (J6step > HighStep)
        {
          HighStep = J6step;
        }

        int J1_PE = 0;
        int J2_PE = 0;
        int J3_PE = 0;
        int J4_PE = 0;
        int J5_PE = 0;
        int J6_PE = 0;

        int J1_SE_1 = 0;
        int J2_SE_1 = 0;
        int J3_SE_1 = 0;
        int J4_SE_1 = 0;
        int J5_SE_1 = 0;
        int J6_SE_1 = 0;

        int J1_SE_2 = 0;
        int J2_SE_2 = 0;
        int J3_SE_2 = 0;
        int J4_SE_2 = 0;
        int J5_SE_2 = 0;
        int J6_SE_2 = 0;

        int J1_LO_1 = 0;
        int J2_LO_1 = 0;
        int J3_LO_1 = 0;
        int J4_LO_1 = 0;
        int J5_LO_1 = 0;
        int J6_LO_1 = 0;

        int J1_LO_2 = 0;
        int J2_LO_2 = 0;
        int J3_LO_2 = 0;
        int J4_LO_2 = 0;
        int J5_LO_2 = 0;
        int J6_LO_2 = 0;
[... 466 lines cut ...]
          else if (highStepCur >= DCCStep)
          {
            delayMicroseconds(DCCSpeed);
            DCCSpeed = DCCSpeed + DCCinc;
          }
          else
          {
            delayMicroseconds(REGSpeed);
          }

          ///////////////////////////////
          /////RESET STEPS AND DELAY ////
          digitalWrite(J1stepPin, HIGH);
          digitalWrite(J2stepPin, HIGH);
          digitalWrite(J3stepPin, HIGH);
          digitalWrite(J4stepPin, HIGH);
          digitalWrite(J5stepPin, HIGH);
          digitalWrite(J6stepPin, HIGH);

          delayMicroseconds(25);

        }
        ////////MOVE COMPLETE///////////
        inData = ""; // Clear recieved buffer
        Serial.print("Move Done");

cqliliping, 2018-01-10:
hi,Chris-Annin.
'PE stands for Pulse Every (loop) and SE stands for Skip Every (loop)' help me a to understand more with it,Maybe  need to make a record when I study the code.
Thanks for your reply!
liliping

closed 2018-03-20 by Chris-Annin