Chris-Annin/AR2 #8: motor driver

Nikolastesla66, 2018-01-15:
Hello Annin I am very interested in your robot.
I want to rebuild the AR2.
Currently I can not find the motor driver board (st-6600) on the internet.
Since I am very new to this field, I do not know what I can take instead. 
Is it possible to install the digital motor drivers?
I would be extremely happy if you answered me.

Chris-Annin, 2018-01-15:
take a look at the manual for the electrical enclosure - this has info for
the digital drivers.  in the latest robot I made I used five DM542T and one
DM320T drivers

Chris-Annin, 2018-03-20:
I eliminated the ST-6600 from the BOM as they are obsolete, please use the DM-542T and DM-320T or any driver of your choice.  doesnt matter too much as long as the micro stepping is set correctly and amperage is correct.

closed 2018-03-20 by Chris-Annin