Chris-Annin/AR2 #9: STEP-files for 3D-printed version

Knechtie, 2018-02-12:
Hi,
could you also provide the STEP files for the 3D-printed version?
I've noticed some differences between the .stl files of the 3D-printed version and the step files (aluminum version, I guess?).

Thank You!

Chris-Annin, 2018-02-12:
which files specifically do you see differences?  I did make minor
adjustments to the .stl files to accomodate for shrink issues especially on
some of the bearing diameters but I left the step files unmodified for use
in machining purposes - are you looking at the small differences for shrink
or are you talking about the different files for the initial design with
ecomonmy series motors and the spring assist.  I am in the process of
updating the design so the printed and aluminum are identical.  I need
to print and put together another printed robot - I am discontiuing use of
the economy motors for J123 and also eliminating the spring I think
depending if the J2 spindle can support load without spring assist but I
need to print and test another robot.  I will try to get this done as soon
as I can and get all the files updated.

Chris

Chris-Annin, 2018-03-20:
email me at chris.annin@gmail.com and I will try to get you the files you need.

closed 2018-03-20 by Chris-Annin

machinekoder, 2018-06-24:
Where can I find the STEP files?