Chris-Annin/AR2 #11: Go To Fine Calibration Position Button

jeffmaeder, 2018-03-18:
The button on the new version 1.4 is not executing a move to position like in the 1.1 software.

Chris-Annin, 2018-03-19:
can you tell me more? which button?

Chris-Annin, 2018-03-19:
I think I see the issue.  im on it.  I'll let you know when I have a fix.

Chris-Annin, 2018-03-20:
posted updated files.  I believe the bug is fixes as I havnt been able to recreate after.  please confirm

closed 2018-03-20 by Chris-Annin

Chris-Annin, 2018-03-21:
> The updated files still seem to have the bug. If I teach the fine calibration position , then jog any axis off the taught position, when I press the move to fine calibration position, nothing happens. This is the position that is stored when I use the teach calibration position button:
>
> Move J [*] X) 628.08 Y) 0.001 Z) 169.769 W) -90.0 P) 90.0 R) -90.0 T) 201.5 Speed-25 Ad 0 As 100 Dd 0 Ds 100 $F

Ok I’ll look again I must be missing something obvious. Thank you for your help and pointing this out.