Chris-Annin/AR2 #12: CAD Files for new Version

BauerPh, 2018-03-18:
Hi,
At first many thanks for your new version.
Would you provide the CAD files for the new 3D printed version?
We need them to built your robot within a school project at a german school for automation technologies.

Thanks in advance.

Chris-Annin, 2018-03-19:
what files do you need?  with the kickstarter trying to launch aluminum
kits i havnt posted the step files. do you need solidworks files or step
files?

BauerPh, 2018-03-19:
solidworks assembly would be great!

BauerPh, 2018-03-19:
And 2 another questions:
- The ATX connector is not required anymore? Because it is still listed in the BOM.
- What is the advantage of the chain option inatead a belt?

Chris-Annin, 2018-03-20:
email me at chris.annin@gmail.com and I will try and get you the files you need.

Chris-Annin, 2018-03-20:
I did switch from ATX to aviation plugs and separate wires as this makes it more practical to make cables of any length you need.  I removed the ATX cable from the design - I just updated the wiring harness manual and removed it from the BOM -thanks for pointing that out. 

The belt was stretching too much so I opted for the chain.

closed 2018-03-20 by Chris-Annin

soswow, 2018-04-07:
@Chris-Annin I am just curious why those CAD files are not part of a repo?