Chris-Annin/AR2 #13: Aluminium part files

modulab, 2018-03-20:
Hi Chris, I've been following your project for some time now and it's quite amazing. 
Recently i finished a metal CNC and i wanted to try a few of the aluminium parts. 
Do you plan to share them ?
thanks,
Paul

Chris-Annin, 2018-03-21:
hello, I did just launch a kickstarter to sell aluminum parts kits
https://www.kickstarter.com/projects/406097865/ar2-6-axis-robot-aluminum-parts-kits
I would like to make sure if I release the models that they are used for
personal use and not to resell.  I will try to get models posted when the
kickstarter id over.  thank you.

Chris

modulab, 2018-03-21:
I understand, Unfortunately i can't pledge for the campaign at the moment.  Have you thought selling the files separately on kickstarter? I assure you it is for experimental use only. 
Good luck with everything!

Chris-Annin, 2018-04-04:
I have added the files as an option on kickstarter.

closed 2018-04-04 by Chris-Annin