Chris-Annin/AR2 #18: Great work, I have a question on 3d print version.

xuanmumy, 2018-11-30:
Hello Chris,
 Thanks for your great job, especially the assemble document, it is very clear. 
 Now I'm planning to make one, I can only make a 3d print version, and I have some questions.
 did you used 3d printed j2 spindle ,j3 spindle and j6 drive post on your plastic version?  
 if so, what material did you used, pla or abs? 
 is it strong enough to use pla? I only have pla filament in hand currently.

aasteven, 2018-11-30:
Hi xuanmumy, 
  I printed all in PLA, but I broke J6 drive post and J3 spindle first because I was doing my first test with calibration and second, the part was infilled at 20% so, I re-printed thoses 2 parts with 80 infill and re-builded the robot arm. I'm still testing. 

For some parts I had to re-print the part smaller 5,10% to fit the berings (Prusa i3 MK3), if not use the dremel to reduce the size.

tks
And thanks again to Chris Annin for all is work.

xuanmumy, 2018-12-07:
> .

really? I am printing these parts at 100% infill.  I will double check my parts' size. I didn't measure them yet. bearings and electrical are on the road.

xuanmumy, 2018-12-07:
> Yes, I 3-D printed all of the spindles, but I did print those at 100% solid out of ABS. I haven’t tried making any of the parts out of PLA so I’m not sure how well that material would hold up.
> […](#)
> Sent from my iPhone

Thanks for reply, I have printed J6 drive by 100% infill, it looks good. hope it can hold on.

closed 2019-07-05 by xuanmumy