Chris-Annin/AR2 #24: 5 axis downgrade

zdenek-onz, 2019-05-30:
Hello, I'm sorry for the stupid question. I'm new here and in control robot arm. It is possible to use this software for a 5-axis robot arm. Or can the 6th axis be disabled in the software? Thank you very much for your answer. Zdenek, Brno, CZ

Chris-Annin, 2019-05-31:
The software is not directly setup for 5 axis control but it will work if
setup correctly - please see this video
https://www.youtube.com/watch?v=pcDfwlcmldE

airgoing, 2020-12-06:
Hi Chris Annin, if the [stepper motor](https://www.motiongoo.com/digital-stepper-driver-10a-42a-20-50vdc-for-nema-17-23-24-stepper-motor) driver can be work for AR2
https://www.motiongoo.com/digital-stepper-driver-10a-42a-20-50vdc-for-nema-17-23-24-stepper-motor
Thanks a lot!