Chris-Annin/AR2 #27: monoprice speaker wire + aviation plugs mystery

i-make-robots, 2019-07-25:
Hi!

I see in the wiring harness documentation it states to get 20 avaiation plugs and 100ft of 18ga speaker wire.  Later in the document it shows where to attach one side of the plugs.  I don't see any mention of the other end or the speaker wire.  What am I missing?  I got similar cable and plugs to put together here and they are proving very hard to assemble. 

Thanks!

i-make-robots, 2019-07-25:
nm, I am blind.  Still... no idea how you did it.  You make it look easy :)

closed 2019-07-25 by i-make-robots

Chris-Annin, 2019-07-25:
6 plugs are installed in small black enclosure on robot (wiring harness).
6 plugs are installed in electrical enclosure (electrical enclosure
manual).  Then 6 extension cables are fabricated (connectors at each end)
to go between the robot (small black enclosure) and the main electrical
enclosure.  Let me know if that makes sense.  Im in the process of
rewriting the manual and consolidating from 3 separate manuals into one
large manual so hopefully there will be less confusion on the aviation
plugs which are called for in each manual.

i-make-robots, 2019-07-26:
I might need a video of your avaiation cable assembly process. We could not get the plug exterior over the cable sleeve to make room while soldering.